adafruit/Adafruit_CircuitPython_EMC2101 #2: Memory error using sample on QT Py

rpavlik, 2021-01-25:
Is there any way that I could slim this library down a bit in terms of runtime memory usage? The code looks pretty complete - anything I can just drop out for testing? I am already using mpy versions of everything (installed via circup), on circuitpython 6.1.0 "haxpress" build (I have the spi flash)

ladyada, 2021-01-25:
you could pull out all the LUT handling stuff,

rpavlik, 2021-01-25:
Oh nifty, I was able to get it to work by commenting out the LUT and the PWM stuff, which still left enough functionality for the example to work (and everything I cared about). Here's the source and mpy. I'll have to look and see sometime if there's an easy-enough way to split this library a little bit so that you get this cut-down version by default, but if you have the RAM you get it all. 
[Adafruit_CircuitPython_EMC2101.zip](https://github.com/adafruit/Adafruit_CircuitPython_EMC2101/files/5868390/Adafruit_CircuitPython_EMC2101.zip)

caternuson, 2021-03-19:
Fixed with #6

closed 2021-03-19 by caternuson