adafruit/Adafruit_CircuitPython_EMC2101 #3: PWM Frequency Bug

MaxBec, 2021-02-07:
Hi guys,

shouldn't this be:

```
    @pwm_frequency.setter
    def pwm_frequency(self, value):
        if value < 0 or value > 0x1F:
            raise AttributeError("pwm_frequency must be from 0-31")
        self._pwm_freq_div = value
```

```
    @pwm_frequency.setter
    def pwm_frequency(self, value):
        if value < 0 or value > 0x1F:
            raise AttributeError("pwm_frequency must be from 0-31")
        self._pwm_freq = value
```

ladyada, 2021-02-07:
ya looks like a copy paste error - wanna try submitting a PR fix? :)

MaxBec, 2021-02-07:
Yes! Will add it!

MaxBec, 2021-02-08:
PR added! :)

closed 2021-02-08 by ladyada

MaxBec, 2021-02-09:
Can you guys make a new release? So that we can use the library? :)

ladyada, 2021-02-09:
hihi we do sweeps a few times a week - @dherrada will be sweeping it!

dherrada, 2021-02-09:
@MaxBec Yep. Just released it.

commit 15d704e9 (2021-02-08): Fix #3
commit c495e121 (2021-02-08): Merge pull request #4 from MaxBec/hotfix_pwm_freq

Fix #3