adafruit/Adafruit_CircuitPython_EMC2101 #12: Typo in conversion rate?

rpavlik, 2021-03-16:
https://github.com/adafruit/Adafruit_CircuitPython_EMC2101/blob/3ae2a7c162a4047ecb0ae55db0adef3d78c84c6d/adafruit_emc2101.py#L236

All the rest of the strings appear to be just a stringification of their value, but the "1/2" is the stringification of "126.0". I think this is a typo but I don't understand this part of the code well enough to be sure.

caternuson, 2021-03-20:
Does look like a typo. Looks like the `ConversionRate` class is currently only being used as a way to validate the `rate`, and that is done based on the existence of the key, which is the 2nd entry of the tuple. So it's probably currently not actually causing any issues, AFAICT.

ladyada, 2021-05-04:
its just a typo - @rpavlik please submit a PR :)