adafruit/Adafruit_CircuitPython_EMC2101 #8: Problem with LUT example on Raspberry Pi 4 (64bit)

asmagill, 2021-03-16:
When I run the example `emc2101_lut_example.py` on a Raspberry Pi 4 with 64bit Raspberry Pi OS (https://www.raspberrypi.org/forums/viewtopic.php?p=1668160, but updated via `apt` to latest available), I get the following error:

~~~sh
amagill@portal:~:%0:$ python3 ./emc2101_lut_example.py 
Traceback (most recent call last):
  File "./emc2101_lut_example.py", line 15, in <module>
    emc.lut[27] = 25
  File "/usr/local/lib/python3.7/dist-packages/adafruit_emc2101.py", line 142, in __setitem__
    self._set_lut(self.lut_values)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_emc2101.py", line 193, in _set_lut
    getattr(self, "_fan_lut_t%d" % (idx + 1)).__set__(self, current_temp)
AttributeError: 'int' object has no attribute '__set__'
~~~

Manual setting (i.e. the simple test) works fine, but I haven't tried the PWM test.

I don't have another Pi 4 running 32 bit to test against, but I'll try on a Pi 3 tonight to see if that makes a difference. Or I might be able to build a new SD card running the 32bit OS and try that on the 4, if I have any currently free -- I'll post my findings, when I get a chance tonight.

rpavlik, 2021-03-16:
There's a bug in the LUT that I literally just figured out and fixed - see #9

asmagill, 2021-03-16:
I'll clone the repository and merge the pull requests in later tonight and let you know how it goes!

asmagill, 2021-03-17:
I can confirm pull #9 fixes my issue, thanks!

caternuson, 2021-03-20:
Yep. Thanks for the fix! Merged it in.

closed 2021-03-20 by caternuson