adafruit/Adafruit_CircuitPython_EMC2101 #21: Using the example LUT example with an actual board does not work

JeremyLaurenson, 2022-02-19:
The example works; but if you remove temp override then the actual board does not actually function with the LUT as it should
See:
```
pi@octopi:~ $ python ./fancontrol.py 
LUT:  20 deg C => 0.0% duty cycle
23 deg C => 20.0% duty cycle
25 deg C => 50.0% duty cycle
28 deg C => 75.0% duty cycle
30 deg C => 100.0% duty cycle
Internal temperature: 25 C
Currently running at 1781.590234 RPM
Internal temperature: 26 C
Currently running at 1779.242175 RPM
Internal temperature: 26 C
Currently running at 1781.590234 RPM
Internal temperature: 26 C
Currently running at 1772.234985 RPM

```

I can take a look at the code next week to see if I spy anything

rivimey, 2022-04-09:
I'm working on this code too, and I think there may be some issues with tach reading, but not sure yet. Did you take another look?

JeremyLaurenson, 2022-04-09:
I actually just implemented it in my own code; so essentially I gave up.

fentonfentonfenton, 2023-07-04:
LUT is only for external temperature I think?